brannondorsey/ProbeKit #6: SSID unicode/character bugs

nbriz, 2015-03-24:
[image]
not entirely sure what's going on when that funky SSID happens, but here's what this particular barf matches up to in the unicode character table ( in case there's any clues ):

 = [Bell](http://unicode-table.com/en/0007/)
 = [End of Transmission](http://unicode-table.com/en/0004/)
 = [Start of Text](http://unicode-table.com/en/0002/)
 = [End of Transmission](http://unicode-table.com/en/0004/)
 = ?
 = [Synchronous Idle](http://unicode-table.com/en/0016/)
2
 = [Backspace](http://unicode-table.com/en/0008/)
 = [Form Feed](http://unicode-table.com/en/000C/)
 = [Device Control Two](http://unicode-table.com/en/0012/)
860H`l
 = [End of Text](http://unicode-table.com/en/0003/)
 = [Start of Heading](http://unicode-table.com/en/0001/)

brannondorsey, 2015-09-10:
One way that we could probably fix this is by checking if all characters in the ssid are within the ASCII (or extended ASCII) range:

```javascript
function isASCII(str, extended) {
    return (extended ? /^[\x00-\xFF]*$/ : /^[\x00-\x7F]*$/).test(str);
}
```

From http://stackoverflow.com/questions/14313183/javascript-regex-how-do-i-check-if-the-string-is-ascii-only

brannondorsey, 2015-09-14:
Here are two examples of how problem packets are printed to the screen via `console.log(...)`:

```
\uFFFD\uFFFD/\uFFFD\u0019$0H`l-\u001Ac\u0000\u0017\uFFFD\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u007F\b\u0004\u0000\b\uFFFD\u0000\u0000\u0000@
```
And
```
xfi\u003Ehtywifi
```

closed 2015-10-19 by brannondorsey