brannondorsey/ProbeKit #7: getFilteredMacs() not returning when filtering by manufacturers

nbriz, 2015-03-24:
i think my filters are working but when u try to filter by manufacturer it deosn't produce butterflies, i think b/c when ur filtering only by manufacturer getFilteredMacs() which runs in applyFilter() is empty even though filter.manufactuer is not, here's a test from my web.dev console:
````javascript
filter.manufacturer
Array [ "60:be:b5:b0:69:f4" ]
applyFilter()
undefined
getFilteredMacs()
Array [  ]
````

brannondorsey, 2015-03-24:
Ah, this is because `filter.manufacturer` does not accept full 17 character MAC addresses. Rather, it expects an array of the first 8 characters (which are manufacturer dependent) of a each MAC, e.g. `60:be:b5`.

nbriz, 2015-03-24:
aaaaaaa.... i forgot we were gonna do it like that... imma edit + re-push ^__^

brannondorsey, 2015-03-24:
I don't know if we actually decided on that, haha that may have been a bad
decision on my part. I'm in the middle of a fix. Whoever gets there first
wins!

brannondorsey, 2015-03-24:
Also, the MAC address itself will never be a filter right (it would only ever show one result)? My change proposes this instead:

[image]

nbriz, 2015-03-24:
fixed && pushed :)

nbriz, 2015-03-24:
ah ok, i pushed changes... its' working like this... but i didn't realize to make the manufacturer the link && not the mac address... imma fix that now

brannondorsey, 2015-03-24:
Awesome! I will ditch my mid-way changes and pull yours as soon as they come in. Thanks!

nbriz, 2015-03-24:
actually... b4 i do this... imma propose something else... so right now it loox nice, when u filter by vendor all the butterflies are different shapes but the same color ( cause vendor dictates color )... however there's the issue that some vendors have more than one set of 00:00:00... ( like xerox and apple ) so my question... when we filter by vendor should it be a strict 00:00:00 or should it instead be filtering by vendor name ( this way it'll include every butterfly regardless of which 00:00:00 they have )... does that make sense?

nbriz, 2015-03-24:
i just pushed changes ( so now the vendor name is the link )... but its still the case that filtering by vendor means strickly filtering by the first 8 chars, so if u find an Apple device and filter by Apple, ur not actually filtering by Apple ur filtering by that specific 8chars ( which belongs to Apple, but there may be other sets of 8chars that also belong to Apple and those get filtered out )

brannondorsey, 2015-03-24:
That does make sense. Lets sort by vendor name not first 8 chars in MAC. The color thing is nice, but I can see it being really unhelpful when trying to check all Apple devices but only returning like 1/8th the results. Plus if there are multiple colored butterflies it might be interesting to learn that that manufacturer has multiple registered MAC prefixes. How does that sound?

Why don't we make `filter.manufacturer` contain an array of strings that of the manufacturer rather than the MAC prefix.

brannondorsey, 2015-03-24:
If you like this idea and integrate it in the front end I can update our filter on the backend to compare these strings.

nbriz, 2015-03-24:
yea that sounds cool, u wanna update filter.manufacturer to take strings + push ur changes + then i'll pull and work on front-end ( its tricky to test front-end when i don't have the right data ...hard to really tell if its working )

brannondorsey, 2015-03-24:
Sure! Will do. Actually, why don't we make `filter.manufacturer=""` instead of `filter.manufacturer=[]` because there can never be a device that is made by more than one manufacturer.

nbriz, 2015-03-24:
ha, very true ^__^

brannondorsey, 2015-03-24:
:+1: Alright, this has been updated in `dataparse.js`.

nbriz, 2015-03-24:
ok... so i pushed changes, but for some reason its not working as it should, when i filter by "Apple" it shows only three butterflies ( different colors, but there's way more than three on the page ) ??? imma be away from my computer for a bit, but maybe u'll notice something?

brannondorsey, 2015-03-24:
Great! I'll try and fix it. Thanks for the updates :balloon:

brannondorsey, 2015-03-25:
Fixed. MACs in `vendorData.json` were stored in all caps while our Tshark values were not. [Using `toUpperCase()` to compare](https://github.com/brannondorsey/ProbeRequestCollectorsKit/commit/8f6c80adc96bab91a5c35359dea1306ffa9d05c2#diff-29b99d91d26c592dc3b49d1d744f2a45R144) did the trick:

closed 2015-03-25 by brannondorsey