brannondorsey/ProbeKit #8: Better data structure for vendorMacs.json

brannondorsey, 2015-03-24:
A few quick thoughts about how we are storing, loading, parsing, and using this small database.

1. We should load this file via AJAX instead of a script include to keep from stalling page load
2. We should store this as valid JSON rather than JS (creating and assigning a `vendor` object).
3. We should use a dictionary-style data structure rather than an array so that lookups are less time consuming:

```json
{
"00:00:00": "Xerox",
"00:00:01": "Xerox",
"00:00:02": "Xerox",
"00:00:03": "Xerox",
"00:00:04": "Xerox"
"00:00:05": "Xerox"
}
```
vs

```json
"mapping": [
      {
         "mac_prefix": "00:00:00",
         "vendor_name": "Xerox"
      },
      {
         "mac_prefix": "00:00:01",
         "vendor_name": "Xerox"
      },
      {
         "mac_prefix": "00:00:02",
         "vendor_name": "Xerox"
      },
      {
         "mac_prefix": "00:00:03",
         "vendor_name": "Xerox"
      },
      {
         "mac_prefix": "00:00:04",
         "vendor_name": "Xerox"
      },
      {
         "mac_prefix": "00:00:05",
         "vendor_name": "Xerox"
      }
]
```

Then lookups are as easy as:

```javascript
if (vendors.hasOwnProperty("00:00:00")) {
    var ven = vendors["00:00:00"]
}
```

rather than

```
for (var i = 0; i < vendor.mapping.length; i++) {
     if (vendor.mapping[i].mac_prefix == "00:00:00"){
        var ven = 	vendor.mapping[i].vendor_name;		    
     }
}
```

This isn't urgent and changes will overhaul some working code, so lets do this another day. I am up to make this change if you agree that this is the right decision @nbriz.

nbriz, 2015-03-24:
yea, that was a lazy implementation on my part ^___^

brannondorsey, 2015-03-28:
Vendor data is now being stored in a hashmap format and loaded via AJAX rather than a `<script>` include tag.

closed 2015-03-28 by brannondorsey