brannondorsey/ProbeKit #10: List View

brannondorsey, 2015-03-28:
It would be helpful to implement a list view to better understand the data being collected as a whole. These views would be a simple list of all networks and devices collected and accessible by clicking Devices or Networks from the stats display (or via another interface).

nbriz, 2015-04-01:
maybe we can make this a modal for network lists && also add manufacturers to bottom left + linst in a modal of its own when clicked

brannondorsey, 2015-04-24:
This would be a nice feature to have for the workshop. Moving this to milestone v0.3.0.

brannondorsey, 2015-04-24:
Network stats now displays all networks in alphabetical order in the modal when clicked. These can all be applied as filters.

closed 2015-04-24 by brannondorsey