brannondorsey/ProbeKit #12: node webkit

nbriz, 2015-04-01:
...mayhaps solution to the standalone app approach

brannondorsey, 2015-04-24:
I've been digging more into Node Webkit and our desire to implement it in a straightforward way without having to alter very much code. I will be creating a `node-webkit` branch to implement these changes. So far here are my thoughts:

- It would be simplest if we could implement `nw` with zero changes to our server/client and websockets model. It looks like this can (and has been) done. Perhaps we can use the `node-main` field in the Node Webkit `package.json` file to launch our server. The file specified by that option acts as a daemon for the life of the app. More about that [here](https://github.com/nwjs/nw.js/wiki/Manifest-format#node-main).

- It is generally discouraged to do that (^), but this wiki page: [About Node.js server side script in nw.js](https://github.com/nwjs/nw.js/wiki/About-Node.js-server-side-script-in-nw.js) actually seems like we might be the edge case where it is a good idea (especially if we still want to have a web-only version of the kit for a real website).

- If the `node-main` plan works out, here are the very small changes I envision us having to make:
    1. Break `server.js` into its own class that can be called with an `options` parameter (basically everything that is now a command line flag.
    2. Create a `nw_start.js` script (launched with `node-main`) that 1) selects a wifi device 2) launches the `Server` class and 3) navigates the app window to `http://localhost:3000`.

brannondorsey, 2015-04-25:
Ok, I've made way more progress on this. Here is a more updated checklist to get things working with Node Webkit

- [ ] Create and manage a `.proberequestcollectorskit/` (or maybe `.probekit`) directory in users `$HOME` directory for mappacks and datapacks. This will keep our download size to a minimum.
- [ ] [Webkit Compatibility](https://github.com/brannondorsey/ProbeRequestCollectorsKit/issues/24) (mainly CSS3 Butterflies)
- [ ] Wireless device selection menu on app start up
- [ ] [Capture privileges](https://github.com/brannondorsey/ProbeRequestCollectorsKit/issues/29) so that user doesn't have to supply root password every time they run the app.
- [ ] Updated install.sh script

closed 2015-05-02 by brannondorsey