brannondorsey/ProbeKit #13: work while u sleep

nbriz, 2015-04-01:
find a way to keep it collecting even if the laptop is shut

brannondorsey, 2015-04-15:
In OSX this is supposed to be achievable with the built in `caffeinate` command (using the `s` option), however I haven't been able to get this functionality to work as described. That command replaced a specific use of the `pmset` command that is still available on Mac. I was able to get the [No Sleep](https://code.google.com/p/macosx-nosleep-extension/) app/plugin to work, which comes with the `NoSleepCtrl` command line interface. This could potentially be installed and used under the hood by our app. I haven't tried anything on Linux, but it seems like there is a very simple [solution here](http://ubuntuforums.org/showthread.php?t=1811403). Also, for further reference, it seems that our desired behavior is casually called "putting a laptop in clamshell mode."

brannondorsey, 2015-05-25:
I don't think that this feature should actually be implemented in our code. Rather, we should point users to 3rd party apps that provide this "No Sleep" functionality in our documentation.

closed 2015-05-25 by brannondorsey