brannondorsey/ProbeKit #16: change math for butterfly size

nbriz, 2015-04-01:
...so that it reflects how many networks it's got

nbriz, 2015-04-19:
got this more or less working

...i think there may be something about our order of operations here that's making it a little buggy.
when a butterfly first appears it's super large ( as though it's knownNetworks length is real large ) when u refresh it goes down to the appropriate size... are we running the makeButteryfly function before updating it's network list?

brannondorsey, 2015-04-21:
Fixed.

brannondorsey, 2015-04-21:
Oops, not actually fixed. Butterflies are now guaranteed to be created created at their smallest size. The issue is that we only have a `makeButterfly()` function, not an `updateButterflySizeBecauseWeReceivedANewNetworkFromThatMacAddress()` function, or perhaps more appropriately named `updateButterfly()` which would change its size. Right now new butterflies are only drawn to the DOM when we receive their first probe, which guarantees they will have only one network at the time.

brannondorsey, 2015-04-22:
Bubble up.

brannondorsey, 2015-04-24:
Fixed with 8c77852. Size transition also naturally has a nice change (^ bubble up).

closed 2015-04-24 by brannondorsey