brannondorsey/ProbeKit #19: add <input/> to filter menu

nbriz, 2015-04-01:
an input field to the filter menu that can:
>> auto complete
>> that prevents u from adding a tag that doesn't exist ( w/some nice visual feedback )
>> has a kind of "smart-tag" functionality, where it knows whether or not the term is a network or a vendor w/out u having to specify this manually

brannondorsey, 2015-04-24:
What is the state of this? Is it something that I can pick up what you are trying to do and start working on it?

nbriz, 2015-04-26:
i was in the middle of the "visual feedback" part
i can finish that up now re-using some of the math from the distanc-butterfly fade && at that point it'll probably be ready for a hand-off to u

nbriz, 2015-04-27:
ok, i more or less finished this
( could use some testing just to be sure )
i've got it auto-completing && adding new filters
...only thing is it only adds networks ( using http://localhost:3000/api/networks ) not vendors
don't think u made an api to search the vendor names right?

closed 2015-04-28 by brannondorsey