brannondorsey/ProbeKit #20: MAPS!!!

nbriz, 2015-04-01:
lets get some maps going... step one, get some beacons

brannondorsey, 2015-04-02:
Been looking a bit into MongoDB with [this pdf](http://m.friendfeed-media.com/ff3dfc8b8bb2d7e3cb5743975ee08f006f4f40cc). Looks super Node.js compatabile + easy/powerful/fun to use. We should look into using it to store our Wigle data scrapes.

brannondorsey, 2015-04-10:
Beacons have now been gotten using [this method](https://github.com/brannondorsey/ProbeRequestCollectorsKit/wiki/Batch-Downloading-Wigle.net-Data). We've got half a million Chicago networks now in a MongoDB database. More soon!

nbriz, 2015-04-19:
ok, so recap on next steps: 
- create an API i can query from a maps page
- API returns ssids + geo + ranking

brannondorsey, 2015-04-21:
API is built and functioning (although it hasn't undergone rigorous testing). You can see a demo at `localhost:3000/api_test.html`.

I haven't implemented ranking because I am not sure if it actually the best thing to do on the back end. It seems like if you make a separate AJAX request for each SSID it should be easy to infer which networks are common vs uniq simply by the number of results returned. The `xfinitywifi` network only takes a moment to load (which is returning 12,000+ results for Chicago) so it shouldn't take to long to just query for everything.

brannondorsey, 2015-04-22:
Wow, we should have just [searched GitHub for "Map Tile Downloader"](https://github.com/search?utf8=%E2%9C%93&q=map+tile+downloader). It looks like there are plenty of good scrappers so that we can use tile data offline. I'm going to look more into this tonight.

nbriz, 2015-04-22:
http://maps.stamen.com/toner/#9/39.9245/-122.5237

nbriz, 2015-04-27:
ok, i've got the maps working locally with leaflet + our custom tiles
we still need to implement some kinda ranking re:how common the network names are
the real popular ones are way too common, proactically plaster all of chicago... need a way to maybe filter those out? ++ also use colors to differentiate common ssids from rare ones

closed 2015-04-28 by brannondorsey