brannondorsey/ProbeKit #25: targeting spinner + info panel

nbriz, 2015-04-22:
randomly targets different butterflies
pulls it's info didactic box >> same as modal except butterfly is realtime copy of the butterly flying in space && includes mini map

nbriz, 2015-04-22:
@brannondorsey >> make me a pair of socks [1] send me macs [2] ask for mac receive networks

brannondorsey, 2015-04-22:
I only got to spend about half of the day so far on Probe Kit stuffs, but I made some good progress :) Here is the jist:

All networks can be grabbed via ajax at

```
http://localhost:3000/api/networks
```

or for the networks from just one device:

```
http://localhost:3000/api/networks?device=MAC
```

but before that, you will probably need some MACS, which can be found at

```
http://localhost:3000/api/devices
```

I decided to make the above "web" accessible rather than served via websockets. That said, new device data needs to be pushed to the installation from the server, so I have made a `newDeviceReceived` websocket event that can be listened for like this:

```javascript
var socket = io.connect("http://" + window.location.host);
	
socket.on('newDeviceReceived', function (mac) {
     // mac is the string representation of a MAC address
});
```

All of this hasn't been too thoroughly tested but should work.

brannondorsey, 2015-04-23:
I've added a new API endpoint to check if a MAC address has a network with a geo location. This is accessible at `http://localhost:3000/api/wigle/hasgeo`.

It expects `device=MAC&collection=COLLECTIONNAME` parameters and returns a `true` or `false` JSON value that can be immediately compared. See example:

```
http://localhost:3000/api/wigle/hasgeo?device=d4:f4:6f:3a:2c:16&collection=wigleChicago
```

Note that you should perform a `result === true` comparison on the data returned as aside from `true` or `false` the API endpoint can also return an `{ error: "some error" }` object.

brannondorsey, 2015-04-23:
[image]

brannondorsey, 2015-04-23:
What do you think about that? With the target cursor animated. You can see a bigger image in `media/`.

nbriz, 2015-04-23:
yea that's nice :)

nbriz, 2015-04-29:
so when i run the server + open the installation view ( with sockets ) often times i can see new devices coming in via the terminal but they don't always show up in the installation ( ie. dont' seem to trigger the sockets event ) ???

so they're actually populating slower than they should it seems

brannondorsey, 2015-04-29:
Hm... I will take a look at this today.

brannondorsey, 2015-04-29:
[image]

`installation_sockets_test.html` appears to be firing the socket even correctly. I am in openLab so there aren't a lot of devices but each one logged to the Chrome console exactly once when it was first shown in the terminal. Are there any notable steps to reproduce the bugs, or is it possible that the problem is in the socket implementation in the `habitat/installation.html` page itself?

closed 2015-09-03 by brannondorsey