brannondorsey/ProbeKit #27: Offline Maps

brannondorsey, 2015-04-23:
I've downloaded zoom levels 0-17 of a 16km region of Chicago and uploaded it to [the releases page](https://github.com/brannondorsey/ProbeRequestCollectorsKit/releases/tag/v0.2.0). Unzip it in `data/maps/tiles` and rename it to `chicago`. Those tiles should then be served by the server at 

```
http://localhost:3000/maps/tiles/chicago/zoom/x/y.png
```

We will probably have to limit the users map zoom/pan so they only view areas that we have map data for.

nbriz, 2015-04-27:
pushed updates to map.html 
its using the local tiles now + leaflet instead of open layers
disregard the comment i made on the other map issue, i forgot that we said we were gonna handle filtering on the client side >> i just pushed changes to map.html >> it now filters out networks that are too common ( ie more than 100 different locations ) + color codes the rest: blue = less than 100, red = less than 10 ( ie. more unique )

brannondorsey, 2015-04-27:
I've changed the local tile layer url from:

```
http://localhost:3000/maps/tiles/chicago
```

to

```
http://localhost:3000/data/maps/tiles/chicago
```

Our static file server serves the data folder at `http://localhost:3000/data` and `http://localhost:3000/maps` was 404ing everything. @nbriz make sure that `http://localhost:3000/data/maps/tiles/chicago` now works for you too.

nbriz, 2015-04-28:
yep, working
just pushed some edits to map.html so that when the butterfly has only common named networks for which there exitsts way too much data instead of seeing an overloaded map ( or empty map as it was ) it says "GPS data may not be accurate"

brannondorsey, 2015-04-28:
Working. Further details can be opened as their own tickets.

closed 2015-04-28 by brannondorsey