brannondorsey/ProbeKit #28: Custom Map Tiles

brannondorsey, 2015-04-24:
[image]

I'm getting comfortable with CartoCSS using Mapbox Studio. Any thoughts on a custom aesthetic? I'm thinking flat and thin lined, but colors?

nbriz, 2015-04-24:
ooo super nice, i'm actually liking those colors already
i've been imagining it very light + minimal so that the butterflies stand out the most + so it doesn't look like we're trying to make a map service ( figure it should just show the basix + as subtle as possible )
what do u think?

brannondorsey, 2015-04-24:
Ok, here is a [live demo](https://api.tiles.mapbox.com/v4/brannondorsey.3b42de50/page.html?access_token=pk.eyJ1IjoiYnJhbm5vbmRvcnNleSIsImEiOiJtMkU2dTJ3In0.GVahfl2rdjG6vVrmFKKghQ#13/41.8561/-87.6263) of the map. I've played a lot with street name size and appearance at different zoom levels. I can make more changes as needed, but if you think it good I can go ahead and download a map pack for it.

nbriz, 2015-04-24:
yea  this is looking really great to me :)

...part of me feels like maybe there's still to much detail in some of the zoom levels... maybe if zoom levels 12 - 15 had more or less the same simplicity as level 11, and we didn't start seeing building shapes + other details till like level 16? ...i don't know how much control that editor gives u + if this is possible or not... not a big deal if u can't +/or if u disagree >> i'd say its ready to rip if u're also feeling good about it :+1:

brannondorsey, 2015-04-24:
Alright great. I'm doing a mappack download of the current state of the map now that we can use. If we feel like there is too much info we can make edits for sure. I was trying to find a good balance of city data that is readable "at a gaze" but not too distracting.

closed 2015-04-28 by brannondorsey