brannondorsey/ProbeKit #44: Mappack + Datapack Cities

brannondorsey, 2015-05-25:
Below is a list of cities that we intend to provide support for. For now all cities are in the USA, however international cities can be included if requested. Please comment here if you would like to request a location.

- [x] Chicago, IL
- [x] Miami, FL
- [ ] New York, NY
- [ ] Los Angeles, CA
- [ ] San Fransisco, CA
- [ ] Houston, TX

brannondorsey, 2015-05-25:
@nbriz feel free to trim and add to this list as you see fit, this was just an initial list based on population.

nbriz, 2015-05-26:
is wigle's data international? if so maybe also cities like Beijing && London ( popular cities which are also heavily surveilled ), also b4 LA ( maybe even b4 NYC ) maybe we should go after the entire valley?

brannondorsey, 2015-10-22:
Closing this as we will now be doing data collections when it makes since for an exhibition/event or when we receive requests.

closed 2015-10-22 by brannondorsey