brannondorsey/ProbeKit #45: How to run an example?

dragonxtek, 2015-08-26:
Hi, I installed ProbeKit, but I tried run it without luck. Could you do a tutorial about how to run Probekit on linux after installed it, to know if I'm doing something wrong? I want to dump wigle information from developing countries.
I am very interested in the project
Thanks

nbriz, 2015-08-26:
hi @dragonxtek :) thnx for the interest >> the project's still pretty raw + undergoing changes but it does work on linux ( i'm running ubuntu ) what xactly do u mean without luck? could u be more specific about errors/issues?

dragonxtek, 2015-08-26:
I want to dump all wifis from a neighbor. I test with nodejs wigle_dump.js
but i couldn't (this js doesn't anything). I run nodejs server.js -i wlan0,
but the page only show 0 networks and don't capture anything.
Do you have a tutorial with the steps to do the correct?
Finally i'm using wigle.js proyect for while, but your project is more
complete, with more options and an interface.

brannondorsey, 2015-09-14:
Hey there, sorry for the late reply. Are you trying to scrape Wigle.net data or simply give our project a go? If it is scraping Wigle.net data then this wiki we created might be helpful for you: https://github.com/brannondorsey/ProbeKit/wiki/Batch-Downloading-Wigle.net-Data

If you would like to run our project let me know and I can send you a pre-release binary.

brannondorsey, 2015-10-22:
Closing this for now. Ping this thread if you would like it to be re-opened.

closed 2015-10-22 by brannondorsey