brannondorsey/ProbeKit #67: Curl not installed by default?

brannondorsey, 2015-10-22:
`install.sh` relies on curl which in one instance of an ubuntu install I found was missing.

JPaulMora, 2016-03-22:
add to install script & wiki entry for said curl-less distro
````
CURL_PRESENT=$(which curl) # or whereis curl
if [ -z  $CURL_PRESENT ]
    then
    echo "Please install curl"
else 
 #curl installed!
echo "Looking good!" 
fi
````

brannondorsey, 2016-03-22:
Thanks for the snippet. I saw that you forked the project :) Were you able to get it running? I don't think this is written anywhere but back when we were actively developing this we were working with the `develop` branch. Also we've got a documentation site here... I should update the repo to reflect that soon: http://probekit.brangerbriz.com.

Cheers!

JPaulMora, 2016-03-22:
Yes, Most stuff worked out just by cloning. Didn't notice there was a develop branch, check ill check that out. By the way, great work over here!