brannondorsey/ProbeKit #70: Map packs instructions

JPaulMora, 2016-03-22:
Would be nice to have a standard way of making map-packs for different cities so work can be distributed.

brannondorsey, 2016-03-22:
Yes, totally agree! I did a write up a while ago about using the little script that I wrote to scrape WiGLE.net data on the wiki [here](https://github.com/brannondorsey/ProbeKit/wiki/Batch-Downloading-Wigle.net-Data). That said, the way that it works is pretty esoteric and hacky. Also, creating an account w/ wiggle gives you only ~10 requests a day. I've had luck emailing Bobzilla from WiGLE.net and gotten my query limit bumped to 200 but I'm not sure if he will do that for everyone. But, if you are feeling adventurous and would like to try downloading data yourself and WiGLE won't up your limit send me an email and I can give you my account credentials.

JPaulMora, 2016-03-22:
I actually got the WiGle scraper working, but for some reason it returns either old data or with a wrong location (haven't figured out which one is it but most probable is location). Because of this I think I'll just ignore their data and come up with a way of importing CSVs directly from my own captures on to the database, this last thing made me post the issue since the Miami.zip didn't appear to have any common structure.

brannondorsey, 2016-03-23:
Hm... the `mappack_miami_25.7818_-80.2551_25km_zoom_0-17.zip` should have structure to it. It is actually a mongodb collection dump that you can import to a mongodb database with [`import_datapack.sh`](https://github.com/brannondorsey/ProbeKit/blob/master/shell/import_datapack.sh).

JPaulMora, 2016-03-24:
The json file imports just fine, but I'm talking about the map "pictures".. I've tried to reverse the way they are arranged in the folders so I could make my own but can't really see a connection between the chunk number and size with its name or the folder where it is.

JPaulMora, 2016-03-29:
Ok, the map packs pictures are in [OpenStreetMap](http://www.openstreetmap.org) order.

brannondorsey, 2016-03-29:
Ah yes, my apologies, I didn't understand your original question. I did in fact use a custom script to batch download tiles that I styled with [TileMill](https://www.mapbox.com/tilemill/).

brannondorsey, 2016-03-29:
Here is the downloader tool that I used for reference: https://github.com/collinprice/Map-Tile-Downloader

JPaulMora, 2016-04-01:
Nice tool, I actually wrote my own once I discovered where the pictures came from (I pushed it to my fork). One thing I still haven't figured out is how can I make the server display the "migration patterns" under the butterflies. It always says they are unavailable. I even faked some probes for ones in miami pack, with no success.